mbs38/yaMBSiavr #4: Usage

akshulg, 2017-12-01:
How to use the library and example code to make hex file for Atmega16 (or any) microcontroller. I copy the code of example.c in Atmel Studios and copy the .h file in include folder in WinAVR. But when I build the code the error shows- "yaMBSiavr.h" No such directory found.

mwue, 2017-12-02:
You have to copy all files together in your project folder, yaMVSiavr.[c,h] and example.c too.

mwue, 2017-12-02:
btw. ... the atmega16 mcu is not supported by the code. Read and modify yaMBSiavr.h to make it work for this mcu. e.g.  mega8 , mega168 , tiny4313 (and some more) are supported yet.

akshulg, 2017-12-02:
But atmega16 is also defined in .h file!

drdbrr, 2017-12-02:
You must modify timer function and interrupt.

mbs38, 2018-03-08:
I suppose that this has already been resolved?

closed 2018-03-08 by mbs38