mbs38/yaMBSiavr #10: buffer overflow when using more registers

kasseller, 2019-10-11:
Hello,

i am now using the lib with atmega644pa, i found the lib has a bug when my definition of input-registers more than 256.

volatile uint16_t inputRegisters[300];

i think the problem is in function "modbusExchangeRegisters"

intToModbusRegister(ptrToInArray+**_(unsigned char )_**(requestedAdr-startAddress),rxbuffer+3,requestedAmount);

is that right?

kasseller, 2019-10-12:
Dear, Max

the problem is not the length of the frame!
if you use (unsigned char), then _**ptrToInArray+(unsigned char )(requestedAdr-startAddress)**_ must be the type of unsigned char by avr studio gcc!

you can do a small test, just read one input-register (e.g. 30260)

mbs38, 2019-10-12:
Oooops. You're right. :-) The cast causes the substraction of requestedAdr and startAddress to be lossy if any high byte of requestedAdr or startAddress is not 0 - which is of course the case with a value like 30260.

Thx for pointing this one out.

mbs38, 2019-10-12:
Fixed in https://github.com/mbs38/yaMBSiavr/commit/db6cbba72ccbb7155e8bdeb03636dbc9adaf4a97

closed 2019-10-13 by kasseller