mbs38/yaMBSiavr #13: Delays and timeouts at different baudrates

uha4, 2020-09-19:
How do i have to set the InterFrameDelays and Char Timeout (yaMBSiavr.h 196-198) for different baudrates?

Just above there is a calculation commented out, that seems do exactly this, but there are different names used. 
And when i do this calculation by hand, i get much smaller values than the used defines.
Can i use this formulas to calculate the values for a different baudrate? Is there a reason you use much bigger values than i get from this?

I'm just thinking about reactivating this calculation. 

In combination with a preprocessor if clause similar to F_CPU testing if BAUD is already set, it would be possible to set the baudrate in the compiler option without touching the library.

mbs38, 2020-09-20:
Hi,

see https://modbus.org/docs/Modbus_over_serial_line_V1.pdf page 14.

"For baud rates greater than 19200 Bps, fixed values for the 2 timers should be used: it is
recommended to use a value of 750µs for the inter-character time-out (t1.5) and a value of 1.750ms for inter-frame delay (t3.5). "

I never needed lower baud rates myself, but what you propose sounds like a good idea.

I will fix that calculation an make its use the default setting.

Best,
Max

uha4, 2020-10-16:
Sorry for the late answer, i expected github to notify me when an answer arrives.

I was changing it in myself, and it seems to work, but i did not test all uses yet.

As i don't have a own repo online to initiate a pull request, here is the diff:

diff --git a/yaMBSiavr.h b/yaMBSiavr.h
index 458d058..f840ac9 100644
--- a/yaMBSiavr.h
+++ b/yaMBSiavr.h
@@ -50,8 +50,10 @@ THE POSSIBILITY OF SUCH DAMAGE.
  *  @author Max Brueggemann www.maxbrueggemann.de
  */
 
-/* define baudrate of modbus */
-#define BAUD 38400L 
+/* define baudrate of modbus if not defined by compiler setting */
+#ifndef BAUD
+  #define BAUD 19200L
+#endif // BAUD
 
 /*
 * Definitions for transceiver enable pin.
@@ -184,18 +186,22 @@ THE POSSIBILITY OF SUCH DAMAGE.
 #define PHYSICAL_TYPE 485 //possible values: 485, 232
 
 /*
-#define modbusBaudrate 38400
-#define modbusBlocksize 10
-#define modbusBlockTime ((float)modbusBlocksize*1000000)/((float) modbusBaudrate) //is 260 for 38400
-#define timerISROccurenceTime 102
-
-#define TimeoutStartOfMessage  (uint16_t)(modbusBlockTime*3.5/(float)timerISROccurenceTime)
-#define TimeoutEndOfMessage (uint16_t)(modbusBlockTime*4/(float)timerISROccurenceTime)
-#define ReceiveMaxGap  (uint16_t)(modbusBlockTime*1.5/(float)timerISROccurenceTime)
+calculating and defining delays and timeout
 */
+#if BAUD <= 19200 // calculate from BAUDRATE
+#define modbusBlocksize 11
+#define modbusBlockTime ((float)modbusBlocksize*1000000)/((float) BAUD) //is 260 for 38400
+#define timerISROccurenceTime 102  // interrupttime in us
+
+#define modbusInterFrameDelayReceiveStart (uint16_t)(modbusBlockTime*3.5/(float)timerISROccurenceTime)
+#define modbusInterFrameDelayReceiveEnd (uint16_t)(modbusBlockTime*4/(float)timerISROccurenceTime)
+#define modbusInterCharTimeout (uint16_t)(modbusBlockTime*1.5/(float)timerISROccurenceTime)
+
+#else  // use fixed times
 #define modbusInterFrameDelayReceiveStart 16
 #define modbusInterFrameDelayReceiveEnd 18
 #define modbusInterCharTimeout 7
+#endif // BAUD
 
 /**
  * @brief    Defines the maximum Modbus frame size accepted by the device. 255 is the default

mbs38, 2021-03-13:
Has been implemented in 2192f5c893505f4985df4c57f7ac725606419c9f

closed 2021-03-13 by mbs38