mbs38/yaMBSiavr #15: About the busmaster can read/write the data amount

wu-bo-yan, 2020-12-31:
Hi dear Max B. 
My name is WU bo-Yan
First of all thank you for your contribution

Read and writ the following data is all normally :
*	coils: 0 to 7
*	discrete inputs: 0 to 7
*	input registers: 0 to 3
*	holding registers: 0 to 3

But my question is 『how to increase my read and write data 』?

for example :
*	coils: 0 to 15
*	discrete inputs: 0 to 15
*	input registers: 0 to 7
*	holding registers: 0 to 7
*
thank you .. Max B.

wu-bo-yan, 2020-12-31:
void modbusGet(void) {
	if (modbusGetBusState() & (1<<ReceiveCompleted))
	{
		switch(rxbuffer[1]) {
			case fcReadCoilStatus: {
				modbusExchangeBits(&outstate,0,『16』);       // I have already tried to adjust this value, 
                                                                                                                but it still not  work
			}
			break;

mbs38, 2020-12-31:
Hi Wu Bo-Yan,
you're welcome.
Did you define outstate according to the new size?
Try 
volatile uint8_t outstate[2];

and
modbusExchangeBits(outstate,0,16);

Best,
Max

wu-bo-yan, 2021-01-04:
Hi dear Max B.
Thanks for the guidance , it get to work !!! 

and i  also adjusted the 『startAddress to modbusDataLocation 』at the following Program , make my modbus startAddress more flexible! 
##uint8_t modbusExchangeRegisters(volatile uint16_t *ptrToInArray, uint16_t startAddress, uint16_t size)
{
	if ((modbusDataLocation>=startAddress) && ((『modbusDataLocation』+size)>=(modbusDataAmount+modbusDataLocation))) ##

##uint8_t modbusExchangeBits(volatile uint8_t *ptrToInArray, uint16_t startAddress, uint16_t size)
{
	if ((modbusDataLocation>=startAddress) && ((『modbusDataLocation』+size)>=(modbusDataAmount+modbusDataLocation)))##

Best Regards
thank you .. Max B.

mbs38, 2021-01-04:
You're welcome.

closed 2021-01-04 by mbs38

mbs38, 2021-01-04:
> and i also adjusted the 『startAddress to modbusDataLocation 』at the following Program , make my modbus startAddress more flexible!

By the way: you should not do that. That check is there to ensure that an array big enough is provided to the modbusExchangeX functions. By removing that check the bus master can read/write into random memory locations :P I doubt that is what you want to achieve. 

What are you trying to do?

wu-bo-yan, 2021-01-05:
Dear Max B.
First of all, the master device I connected is HMI, and the slave device is AVR-mega2560, and the start Address of each screen of the HMI is different, so I must have a more flexible start Address checking mechanism
E.g:
Screen. 1 is 01 01 **00 00** 00 04 3D C9
Screen. 2 is 01 01 **00 0A** 00 02 9D C9
Screen. 3 is 01 01 **00 0E** 00 0B 1C 0E
Screen. 4 is 01 01 **00 1A** 00 06 9D CF

mbs38, 2021-01-09:
Modbus does not support multiple master! You will always get collisions by doing this.