mbs38/yaMBSiavr #16: Cant get value from slave on each request

abhisekmanna, 2021-08-20:
Hi, 
I am using Atmega1284p. My baud rate is 9600, f_cpu set on 8mhz. I want to read holding registers of my rtu slave device. I was able to get data from slave after 15-20 times request on loop and after that I am getting response. I want to get response data from slave whenever I request for data from slave. Please help.

mbs38, 2021-08-21:
Hi, you're welcome.

Without having seen your code I can't help.

Also, as stated in the master-example.c, the library is meant to be used as slave device only (hence the name). Maybe some changes in the last two years have broken the hacky usage as a master. Fixing that though has zero priority for me atm.

Best,
Max

mbs38, 2021-09-14:
Can't confirm this atm.

closed 2021-09-14 by mbs38