mbs38/yaMBSiavr #17: Trying to compile

Gaitashaper, 2021-09-07:
Thank you very much for your code, is a interesting work!!

I am trying to compile the example code but I can not doing, I have a lot of error during compiling process.

I am using AVR Studio 6 and example code same as posted here

What compiling software you are using?

Thank you very much in advance

mbs38, 2021-09-07:
Hi, 
I am using makefiles + avr-gcc.

What errors do you get?

Best,
Max

Gaitashaper, 2021-09-07:
Hi Max, thanks for your answer. Next the image of AVR Studio Errors. I hope you can help me. Thank you very much in advance.

[image]

mbs38, 2021-09-07:
Hi,
AVR Studio seems to be a little more pedantic and I have forgotten to put an "extern" in front of the definitions of DataPos and rxbuffer within the header file. I've pushed a fix (ea26b7b). Hope this fixes the issue for you.

Best,
Max

Gaitashaper, 2021-09-07:
Hi Max, thanks for your answer and cooperation. Now, that is a situation........ What do you think about?

[image]

mbs38, 2021-09-07:
Looks like yaMBSiavr.c is loaded in multiple places. Did you change the example.c in any way?

Gaitashaper, 2021-09-09:
Hi Max, I was able to compile the program with the extern added. I was making a mistake during compilation. The program is ok and compiling well in AVR Studio. Today I am goin to test it on the hardware. Thank you very much for your answer, support and cooperation. Sebastián.

mbs38, 2021-09-14:
Good to hear that you have been able to resolve the issues you had been experiencing. 
You're welcome.
Max

closed 2021-09-14 by mbs38