mbs38/yaMBSiavr #18: Arduino Library...?

gicking, 2021-11-20:
hi Max,

I really like your Modbus library, as it just does the jobs, no frills...  :-)   To increase visibility, have you considered making it available as an Arduino library? Basically all you need to do is to change your folder structure and add 2 description files. 

PS: my STM8 port of your library can be found [here](https://github.com/gicking/STM8_headers/tree/master/examples/Modbus_RTU). Thanks again!

mbs38, 2021-12-20:
Hi Georg,

thx for your message. Highly appreciated. I have been asked several times to bring the lib to Arduino but I never really use Arduino so this is extremely low on my list. So low in fact that I will probably never do it. If someone else did it I would be more than happy.

Thanks for pointing me to your STM8 port. Very nicely done. I will probably use it. I've been thinking about porting to STM for a while :-)

Best,
Max

gicking, 2021-12-22:
Hi Max,

:-)  On 2nd thought porting _yaMBSiavr_ to Arduino would probably be pretty awkward, because the Arduino framework does not support to attach functions to Uart interrupts out of the box. While lib [_NeoHWSerial_](https://github.com/SlashDevin/NeoHWSerial) adds this for the 8-bit AVR boards, this is not possible for the newer 32-bit ARM, e.g. Due (see [this discussion](https://github.com/arduino/ArduinoCore-sam/pull/95)), let alone the plethora of "Arduino-like" boards.

I believe that your implementation is correct, and that handling Modbus via polling is not the way to go. However, I did find  several Arduino libraries which seem to do a pretty good job **if** the protocol handler is called every 1ms or so. So please ignore my proposal, and let's keep your lib as it is, i.e. for bare-metal programmers.

As for my STM8 port, please feel free to use it! How about you add a "port" folder to your repo for different kinds of controllers? 

In any case thanks again for your effort! Well done...

Regards, Georg

mbs38, 2023-12-31:
Hi,
sorry for the late response. Thx for pointing me to your STM8 port! :-)

Best regards,
Max

closed 2023-12-31 by mbs38