mbs38/yaMBSiavr #21: Timer0 configuration for 8MHz clock

clickworkorange, 2023-12-27:
Thank you for this nice little library! I've built [a simple RTU](https://github.com/clickworkorange/AVR-PWM-RTU) using this library which works perfectly in a simulator running at 2MHz, but I'm having communication issues when running it on an actual ATMega328 clocked at 8MHz (using an external crystal). I need the higher clock to get sufficiently fast PWM outputs (using Timer1 & Timer2). The RTU responds correctly to commands at 9600 baud but the replies are not valid due to some timing issue. I understand that the modbusTickTimer ISR expects a 100us clock from Timer0, which works at 2MHz and prescaler 8. Unfortunately the next higher prescaler for Timer0 is 64, while 8MHz operation would need this to be 32. Is there another parameter I can adjust to get the correct timing?

mbs38, 2023-12-27:
Hi,
the timing of the UART should be fine when using an 8 MHz crystal (0.2 % error). I would recommend using a baud rate crystal though (like 7.3728 MHz). I usually go for the highest frequency baud rate crystal possible. With the 328p @ 5V I would go for 18.432 MHz.

About your issue:

You're probably switching the enable pin off in every loop with line 93:
[image]

Also: When fixing that keep in mind that an operation like PORTD|= ... can be interrupted and if the ISR also accesses PORTD (like the ISRs in the modbus library do!) you will get into terrible trouble.

clickworkorange, 2023-12-27:
Ah. Thank you! I forgot I was forcing the ports in the setLevel function, that's rather crude. But it shouldn't be called in the main loop, only when a button is pressed or when function code 10 is received. Perhaps that's what you meant. I had only been testing the Modbus communication by doing just that, and since FC 10 loops through the available channels it will drop into case 2 and reset the enable pin. I guess the 50us delay before it going low was spent dealing with cases 0 and 1. Doh! I can confirm that function code 3 (read multiple registers) works flawlessly - if only I had tested that I might have found the issue on my own (note to self: try more than one thing next time). 

I'm not sure I understand what you mean with the comment about port assignments being interruptible, I always thought these were monolithic - as in the byte being transferred to the port in a single step. Do you mean it might get half way through setting the port value before getting interrupted? That would indeed cause some major head scratching - particularly for a novice like me! Begs the question - though I know this is not the best place for it - how should I go about setting pins on a port which has some pins controlled in an ISR?

clickworkorange, 2023-12-28:
Closing this since it was a wetware error unrelated to yaMBSiavr - with thanks to @mbs38 for clearing it up!

closed 2023-12-28 by clickworkorange

mbs38, 2023-12-28:
You're welcome. 

> Edit: The answer might be as simple as disabling interrupts at the top of the function, and re-enabling them at the end.

Then you will lose interrupts in that time. But the thought is not bad at all. 

```
#include <util/atomic.h>
...
...
ATOMIC_BLOCK(ATOMIC_FORCEON) {
    PORTD|=0x01;
}
```

That is all you need.